yasheena/telnetspy #1: It is possible to use it without the ring buffer?

zoomx, 2018-06-21:
It is possible to use it without the ring buffer?

I mean: if I sometimes want to connect to the ESP8266 to se the actual debug messages and not the previous ones? Just like serial, if there is one connected sees the messages otherwise they are lost.

yasheena, 2018-06-22:
I have not testet it, but you can change TELNETSPY_BUFFER_LEN in TelnetSpy.h to 0. This should switch off the buffer. If it does not work, then try using 1. This would store only the last char.

zoomx, 2018-06-22:
I will test it in the next days, thanks.
It compiles but I had to fix a bug (maybe): in TelnetSpycpp files I had to substitute the only two max() with _max() as described here
[https://github.com/esp8266/Arduino/issues/2073](url)
The bug appears also with unmodified library.

zoomx, 2018-06-22:
If I set the buffer to 0 I can connecti with telnet but get a Fatal exception, I can read the first lines on telnet terminal.
Setting buffer to 1 seems to work, thanks!

yasheena, 2018-09-08:
I just released a new version. Now you can set the buffer size also to 0 to disable the buffer. Also the RingBuffer extension is no longer needed, because I implemented my own ring buffer. May be the new function setStoreOffline(bool store) is interesting for you: You can dynamically switch of the data storing if there is no telnet connection. So its possible to store only important messages in the buffer, if there is no connection. The unimportant data will be transferred only when a client is connected.

closed 2022-06-18 by yasheena