yasheena/telnetspy #2: Possible bug in handle() method...

eruivo, 2018-08-23:
Hi,
First of all thank you for your library, it is just what I was looking for to incorporate in one of my projects.
I have however detected that if an active connection from a client happens to go down before or without properly disconnecting the server will always drop any new connection attempt.

    **if (!started || serverClient) {
      telnetServer.available().stop();**
    } else {
      serverClient = telnetServer.available();
      if (strlen(welcomeMsg) > 0) {
       serverClient.write((const uint8_t*)welcomeMsg, strlen(welcomeMsg));
       serverClient.flush();
      }
    }

For now I have just made a change to stop/disconnect any active connection if a new one comes in but I think implementing a simple watchdog timeout on active data transfer to stop/disconnect would be a better solution.
If the intended behaviour is as it currently is, please forgive me and ignore this.

yasheena, 2018-09-08:
Hi,
this is not a bug but a problem. If there is no traffic on a telnet connection the server has problems to detect a disconnected client. It will happen but it needs a very long time. To get rid of this problem I implemented a ping function which sends periodically a null character to the telnet client if there is just no traffic. So a disconnected client will be detected very fast. This should solve your problem.
I did also a lot of other extensions/improvements (see README.md), so the external RingBuffer component is no longer neccessary. But you have to add
`TelnetSpy SerialAndTelnet;`
to the beginnig of your script, because it is no longer globally defined. So its now also possible to create the TelnetSpy object only if it is neccessary.

eruivo, 2018-09-25:
Hi,
Thank you for your reply and time to look into the situation and I'm sorry for referring to it as a Bug.
I also will be checking out your latest version and the new features, I particularly like the fact that you have eliminated some of the external dependencies and the need for a global definition.

closed 2022-06-18 by yasheena