yasheena/telnetspy #4: disconnect telnet client feature?

proddy, 2019-01-01:
One idea to add is a function to close a telnet client session. For example in an active Telnet session, I would capture a ctrl-D (from a `SerialAndTelnet.read()`) which would exit the Telnet session, but keep the serial and everything else active. The only way I think you could do this now is by calling end() which kills both the server and client? 

so like

```
void TelnetSpy::disconnectClient() {
    if (client.connected()) {
        client.flush();
        client.stop();
    }
    if (connected && (callbackDisconnect != NULL)) {
        callbackDisconnect();
    }
    connected = false;
}

arezi, 2019-03-15:
it would be great! :)

mkh595, 2019-12-13:
> The only way I think you could do this now is by calling end() which kills both the server and client?

You can call ```SerialAndTelnet.setPort()``` and this will restart only telnet session.

My application already use HTTP client so telnet session is never ended when telnet client is closed.

yasheena, 2022-06-23:
Thank you for your hint. Added for version 1.2.

closed 2022-06-23 by yasheena