yasheena/telnetspy #13: Example code doesn't comile on 1.8.13

AverageGuy, 2020-06-30:
/home/jwl/sketchbook/libraries/telnetspy-master/TelnetSpy.cpp: In member function 'void TelnetSpy::setDebugOutput(bool)':
/home/jwl/sketchbook/libraries/telnetspy-master/TelnetSpy.cpp:406:42: error: invalid conversion from 'void*' to 'fp_putc_t {aka void (*)(char)}' [-fpermissive]
   os_install_putc1((void*) TelnetSpy_putc);  // Set system printing (os_printf) to TelnetSpy
                                          ^
In file included from /home/jwl/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/include/os_type.h:28:0,
                 from /home/jwl/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/include/user_interface.h:28,
                 from /home/jwl/sketchbook/libraries/telnetspy-master/TelnetSpy.cpp:12:
/home/jwl/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/include/ets_sys.h:219:6: error:   initializing argument 1 of 'void ets_install_putc1(fp_putc_t)' [-fpermissive]
 void ets_install_putc1(fp_putc_t routine);
      ^
/home/jwl/sketchbook/libraries/telnetspy-master/TelnetSpy.cpp:415:50: error: invalid conversion from 'void*' to 'fp_putc_t {aka void (*)(char)}' [-fpermissive]
    os_install_putc1((void*) TelnetSpy_ignore_putc); // Ignore system printing
                                                  ^
In file included from /home/jwl/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/include/os_type.h:28:0,
                 from /home/jwl/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/include/user_interface.h:28,
                 from /home/jwl/sketchbook/libraries/telnetspy-master/TelnetSpy.cpp:12:
/home/jwl/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/include/ets_sys.h:219:6: error:   initializing argument 1 of 'void ets_install_putc1(fp_putc_t)' [-fpermissive]
 void ets_install_putc1(fp_putc_t routine);
      ^
Multiple libraries were found for "ArduinoOTA.h"
 Used: /home/jwl/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ArduinoOTA
 Not used: /home/jwl/sketchbook/libraries/ArduinoOTA
exit status 1
Error compiling for board Generic ESP8266 Module.

AverageGuy, 2020-07-01:
So I decided to hack away at the problem and came up with this solution.  It now compiles but I havn't tested it to see if it works.  The compile errors go away:
--- TelnetSpy.cpp.orig	2019-06-11 03:39:16.000000000 -0400
+++ TelnetSpy.cpp	2020-07-01 08:28:32.772613248 -0400
@@ -403,7 +403,7 @@
 	if (debugOutput) {
 		actualObject = this;
 #ifdef ESP8266		
-		os_install_putc1((void*) TelnetSpy_putc);  // Set system printing (os_printf) to TelnetSpy
+		os_install_putc1( TelnetSpy_putc);  // Set system printing (os_printf) to TelnetSpy
 		system_set_os_print(true);
 #else // ESP32
 		// ToDo: How can be done this for ESP32 ?
@@ -412,7 +412,7 @@
 		if (actualObject == this) {
 #ifdef ESP8266		
 			system_set_os_print(false);
-			os_install_putc1((void*) TelnetSpy_ignore_putc); // Ignore system printing
+			os_install_putc1( TelnetSpy_ignore_putc); // Ignore system printing
 #else // ESP32
 			// ToDo: How can be done this for ESP32 ?
 #endif

Results from compile:

Executable segment sizes:
RODATA : 1540  ) / 81920 - constants             (global, static) in RAM/HEAP 
DATA   : 1276  )         - initialized variables (global, static) in RAM/HEAP 
IRAM   : 27856   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
BSS    : 25576 )         - zeroed variables      (global, static) in RAM/HEAP 
IROM   : 285660          - code in flash         (default or ICACHE_FLASH_ATTR) 
Sketch uses 316332 bytes (33%) of program storage space. Maximum is 958448 bytes.
Global variables use 28392 bytes (34%) of dynamic memory, leaving 53528 bytes for local variables. Maximum is 81920 bytes.

sebitnt, 2021-02-19:
After applying the above changes it is working fine for me. The author should update the code. Thank you!

closed 2022-06-21 by yasheena